MakerBay/Coral_Reef_Mapping_Drone #39: CAD Thruster mounting sub assembly

MikeyBenori, 2020-04-08:
based on sketch model / prototype add this design to master CAD file

MikeyBenori, 2020-04-09:
[image]

MikeyBenori, 2020-04-09:
Here is the link to the current CAD files
https://github.com/MakerBay/Coral_Reef_Mapping_Drone/tree/master/1_Mechanical

MikeyBenori, 2020-04-09:
Design task overview
-Assembly of: 2 x  Thrusters + 2 fins, + mounting board, seaweed / debris cages

Design Requirements 
- Think about the every components attaches together - screws, joints etc 
- Think about the assembly attached to the main frame of the boat
- ensure the sub assembly height adjustable for lowering the thrusters 
- design for material (plastic, or aluminum) 
-doesn't need to be full board across

pcruickshank, 2020-04-11:
@MikeyBenori - here is a first pass concept model of a sub-assembly to hold the thrusters + seaweed cages. A few things to note:

- I modeled wiring from the main underwater cylinder to the propellers to show how to avoid twisting these around the tubing
- Per our discussion, I've sketched out the sub-assembly as a C-shaped board looping around the back, with a connecting bar to the main tubing. With this design, the C-shape is not tangled with the wiring from the propellers or other cables (which may happen if the bar is crossing inside the tubing cage). If the sub-assembly needs to be detached, this can be done in one step by detaching the C-board from the connecting bar, or the connecting bar from the tubing cage.
- The C-board can be flipped 180 degrees so that it is facing backwards, and the propellers can be placed further back from the tubing cage.
- The height of the connector from the C-board to the tubing is not adjustable at the moment. However, I think it worth exploring this idea a bit more - is the goal to be able to physically adjust the propeller height in between uses? I could see this as being achievable through longer screws. Let's brainstorm on this.
- The instructions mention the fins, but we really didn't discuss that much in person. As such, they are not in this model. Nonetheless, they could be attached to the C-board closer to the middle. I can add those in v2 (though the fins included in the CAD file are quite large relative to other parts of the model)
- Given that the thrust force will be coming from the propellers, I think it is worth iterating on how to best connect the sub-assembly to the main tubing. This connector could be a weak point and may need to be reinforced / redesigned so that the sub-assembly doesn't move vis-a-vis the main cage while in use.

Thanks!

[image]
[image]
[image]
[image]

pcruickshank, 2020-04-20:
In last week's meeting, feedback on the sub-assembly concept design focused on 1) making it more firmly connected to the main assembly, 2) experimenting with a model where the propellers are connected on each side rather than via one sub-assembly in the middle, and 3) adding in fin(s)/blade(s). Based on this feedback, I've developed 4 different concept models. Each follow in a separate post below.

pcruickshank, 2020-04-20:
Model 1 extends the connecting bar that connects the sub-assembly to the tubing frame. This connecting bar is stabilized via screws at two points on the frame. Additionally, where the connecting bar connects with the sub-assembly, a single fin has been added on the underside.
[image]
[image]
[image]
[image]

pcruickshank, 2020-04-20:
Model 2 has the subassembly attached at three points to the top of the tubing frame. The sub-assembly drapes over each side where the propellers and fins are connected. The fin is in front of the propeller. The side hanging part of the sub assembly also can be adjusted up or down (to change the height of the fin + propeller) by changing which level the set of three screws are set.

[image]
[image]
[image]
[image]

pcruickshank, 2020-04-20:
Model 3 is a variation on Model 2. In Model 3, a single fin is affixed behind the tubing frame. It is attached to the sub assembly via a connecting bar that extends on the top back half of the tubing. The fin is not adjustable. However, the height of the propellers is still adjustable in the same fashion as Model 2.

[image]
[image]
[image]
[image]

pcruickshank, 2020-04-20:
Model 4 is a different variation on Model 2. The only difference between 2 and 4 is that the sub-assembly does not have a over-arching connecting plate on the top of the tubing frame. In this model, the sub-assembly is actually two separate parts, each connected to the right and left side tubing, respectively. Fins are in front of each propeller. Each right and left side sub-assembly are detachable, and each allows for adjustment of the height of the propeller + fin.

[image]
[image]
[image]
[image]